hallard/WeMos-TIC #7: soldering schema for WeMos Mini S2 ?

mcarbonneaux, 2022-09-18:
how to solder your adapter to the mini s2 ? use external line or internal ?
you have made the schematique for the esp32 mini dev, but not for the other one !
- https://github.com/hallard/WeMos-TIC/raw/master/pictures/WeMos-TIC-assembled-bot.png 
- https://github.com/hallard/WeMos-TIC/raw/master/pictures/WeMos-TIC-assembled-top.png

[image]

when see you table her : https://github.com/hallard/WeMos-TIC#detailed-description

hallard, 2022-09-20:
If you look at the schematic of the shield (which is the same for all module), Teleinfo RX is connected to `GPIO13` (to leave RX/TX is for serial console) so it's `GPIO11` on S2 and all magic match :-)

Blank text on the PCB (silk) is the one for ESP8266 that may, I must admit do some confusion, but that's the price of compatibility and this is why I put mapping table on the readme.

closed 2022-09-20 by hallard

mcarbonneaux, 2022-09-21:
[image]&nbsp;
[image]&nbsp;
[image]&nbsp;
[image]&nbsp;

[image]
[image]
[image]
[image]

hallard, 2022-09-21:
Thanks for sharing