hallard/WeMos-TIC #14: Portage vers Micropython

mdaeron, 2022-11-12:
Bonjour, merci pour ce projet !

Je suis un grand fan du Wemos S2 Mini que j'utilise sous Micropython pour pas mal de projets. J'aimerais acheter ce module et écrire moi-même (sous license libre évidemment) le code permettant de lire/sauvegarder/transmettre les valeurs de consommation. J'ai un peu regardé ici mais pour je n'ai pas encore compris si le code "habituel" (Tasmota Téléinfo ?) est open-source, et si oui où on peut le trouver.

hallard, 2022-11-12:
Le mieux (et le plus simple) et peut être de prendre le code de la librairie [LibTeleinfo](https://github.com/hallard/LibTeleinfo) (celle ci est incluse dans tasmota)

mdaeron, 2022-11-12:
Merci ! Du coup je viens d'en commander un sur Tindie. Beau projet bien ficelé, merci aussi pour les infos sur le blog.

closed 2022-11-12 by mdaeron