hallard/WeMos-TIC #15: Wemos-Teleinfo & ESPHome: "bad crc" and "no group found" errors

tomav, 2023-01-17:
Hi @hallard, thanks for the wemos teleinfo board, I got mine from Tindie.
It works but I got these errors:

```
[23:12:12][E][teleinfo:133]: No group found
[23:12:16][D][sensor:127]: 'WiFi Signal dB': Sending state -61.00000 dBm with 0 decimals of accuracy
[23:12:43][E][teleinfo:038]: bad crc: got 60 except 56
[23:12:43][E][teleinfo:038]: bad crc: got 60 except 50
[23:12:43][D][sensor:127]: 'Linky Index': Sending state 35917.92578 kWh with 0 decimals of accuracy
[23:12:43][E][teleinfo:038]: bad crc: got 34 except 41
[23:12:43][E][teleinfo:168]: Invalid value for tag EASF02
[23:12:43][E][teleinfo:038]: bad crc: got 41 except 72
[23:12:43][E][teleinfo:038]: bad crc: got 41 except 35
[23:12:43][E][teleinfo:038]: bad crc: got 32 except 88
[23:12:43][E][teleinfo:038]: bad crc: got 32 except 95
[23:12:43][D][sensor:127]: 'Linky Puissance Souscrite': Sending state 12000.00000 VA with 0 decimals of accuracy
[23:12:43][D][sensor:127]: 'Linky Puissance Apparente': Sending state 2466.00000 VA with 0 decimals of accuracy
[23:12:43][E][teleinfo:038]: bad crc: got 94 except 84
[23:12:43][E][teleinfo:038]: bad crc: got 24 except 77
[23:12:43][E][teleinfo:038]: bad crc: got 60 except 86
[23:12:43][E][teleinfo:038]: bad crc: got 57 except 32
[23:12:55][D][sensor:127]: 'Uptime': Sending state 1532.19702 s with 0 decimals of accuracy
[23:13:16][D][sensor:127]: 'WiFi Signal dB': Sending state -61.00000 dBm with 0 decimals of accuracy
[23:13:43][E][teleinfo:038]: bad crc: got 60 except 87
[23:13:43][D][sensor:127]: 'Linky Index': Sending state 35917.97656 kWh with 0 decimals of accuracy
[23:13:43][E][teleinfo:038]: bad crc: got 36 except 57
[23:13:43][E][teleinfo:038]: bad crc: got 36 except 95
[23:13:43][E][teleinfo:038]: bad crc: got 41 except 77
[23:13:43][E][teleinfo:038]: bad crc: got 41 except 40
[23:13:43][E][teleinfo:038]: bad crc: got 32 except 87
[23:13:43][E][teleinfo:038]: bad crc: got 32 except 33
[23:13:43][D][sensor:127]: 'Linky Puissance Apparente': Sending state 2957.00000 VA with 0 decimals of accuracy
[23:13:43][E][teleinfo:038]: bad crc: got 94 except 89
[23:13:43][E][teleinfo:038]: bad crc: got 94 except 49
[23:13:55][D][sensor:127]: 'Uptime': Sending state 1592.19702 s with 0 decimals of accuracy
```

Same in historical or standard mode.
I use ESPHome, with an ESP32 (lolin s2 mini).

Here's my config, nothing too fancy.

```
uart:
  id: uart_a
  rx_pin: GPIO11
  baud_rate: 9600 # historical => 1200 / standard => 9600 
  parity: EVEN
  data_bits: 7

teleinfo:
  id: myteleinfo
  uart_id: uart_a
  update_interval: 30s
  historical_mode: false # set to false when standard mode enabled

#### STANDARD

 # Energie Active soutirée totale
  - platform: teleinfo
    id: hc_hp
    tag_name: "EAST"
    name: "Linky Index"
    unit_of_measurement: "kWh"
    icon: mdi:flash
    teleinfo_id: myteleinfo
    device_class: "energy"
    state_class: "total_increasing"  
    filters:
      - filter_out: 0.0
      - lambda: |-
          return x/1000;
    
 # Puissance apparente de coupure
  - platform: teleinfo
    id: pcoup
    tag_name: "PCOUP"
    name: "Linky Puissance Souscrite"
    unit_of_measurement: "VA"
    icon: mdi:flash
    teleinfo_id: myteleinfo
    filters:
      - filter_out: 0.0
      - lambda: |-
          return x*1000;

# Puissance apparente instantanée ph1 
  - platform: teleinfo
    id: papp
    tag_name: "SINSTS"
    name: "Linky Puissance Apparente"
    unit_of_measurement: "VA"
    device_class: "power"
    state_class: "measurement"  
    icon: mdi:flash
    teleinfo_id: myteleinfo

# Tension moyenne ph1
  - platform: teleinfo
    id: umoy1
    tag_name: "UMOY1"
    name: "Linky Umoy"
    unit_of_measurement: "V"
    icon: mdi:flash
    teleinfo_id: myteleinfo

# Courant efficace ph1
  - platform: teleinfo
    id: iinst
    tag_name: "IRMS1"
    name: "Linky Intensité"
    unit_of_measurement: "A"
    icon: mdi:flash
    teleinfo_id: myteleinfo

#### / STANDARD
```

Any idea of what could be the issue?
Any way to check if it's the teleinfo esphome module or the hardware?
Same kind of errors were observed here: https://github.com/esphome/issues/issues/2060#issuecomment-882589485

Thanks for your help!

yannpub, 2023-01-18:
I am running on an esp8266, you can have a look at my config here: https://community.ch2i.eu/topic/1369/wemos-teleinfo-et-esp-home.
it will require small adjustment for esp32, but from what I've read in the past, issues can occur with the uart bus, or the GPIO you chose.

hallard, 2023-02-27:
You can check hardware with some examples of [LibTeleinfo](https://github.com/hallard/LibTeleinfo/tree/master/examples) such as passthru or stats.

closed 2023-02-27 by hallard

harrygr, 2024-03-04:
@tomav did you ever solve this? I have a very similar setup to you (Board off Tindie, ESP32 S2 mini), but my Linky is Historical mode. I'm getting the same errors:

```
[E][teleinfo:038]: bad crc: got 72 except 32
[E][teleinfo:038]: bad crc: got 44 except 60
```

tomav, 2024-03-11:
Hi @harrygr, sadly nope, so I moved quickly to https://github.com/fairecasoimeme/Zlinky_TIC
Works great with MQTT, and nice form factor, no need of an external 5v.

harrygr, 2024-03-12:
@tomav thanks for the response. I did get it working in the end, albeit with still a few `bad crc` errors. However the metrics I'm interested in are getting recorded and injested by HomeAssistant.

Turns out I had to carefully read the [Teleinfo datasheet](https://www.enedis.fr/media/2035/download) to figure out which tag names work for my Linky setup. I ended up with the following (for single-phase, single rate, historical):

```
sensor:
  - platform: teleinfo
    tag_name: "BASE"
    name: "Usage"
    device_class: energy
    unit_of_measurement: "kWh"
    state_class: "total_increasing"
    accuracy_decimals: 3
    icon: mdi:gauge
    filters:
      - multiply: 0.001 
      - median:
          window_size: 5
          send_every: 4
    teleinfo_id: linky
  - platform: teleinfo
    tag_name: "IINST"
    name: "Current"
    unit_of_measurement: "A"
    icon: mdi:current-ac
    teleinfo_id: linky
  - platform: teleinfo
    tag_name: "PAPP"
    name: "Apparant Power"
    unit_of_measurement: "VA"
    icon: mdi:flash
    teleinfo_id: linky
```