hallard/WeMos-TIC #17: Niveau de signal insuffisant

Gixy31, 2024-07-19:
Bonjour,
Je viens de recevoir ma carte Wemos TeleInfo, et une fois mise en place j'ai le souci suivant:
Mon oscilloscope décode parfaitement les trames à la sortie de la carte, mais pas mon ESP32. Après avoir vérifié mon soft, j'ai vu surf l'oscillo que le signal de sortie faisait à peine plus de 1V, ce qui explique que l'ESP ne le voit pas. Pour confirmer, j'ai amplifié le signal de sortie avec un transistor et là tout va bien (en passant en mode inverse à cause du transistor), j'ai bien du 3,3V d'amplitude. 
Je suspecte que c'est le MOSFET qui est défectueux, qu'en pensez-vous ?

hallard, 2024-07-19:
Bonjour,

Quel type de compteur, quel mode de la téléinfo ? si Linky Standard avez vous essayé de bouger le petit potentiomètre de réglage de sensibilité ?

Gixy31, 2024-07-19:
Bravo pour la réactivité !
Oui, Linky en mode historique. Le potentiomètre ne fait que changer la "propreté" du signal, pas son amplitude, ce qui est normal justement avec l'isolation galvanique. Au début je suspectais un pont diviseur parasite avec la diode bleue, mais il ne peut pas y avoir de courant inverse donc ce n'est pas ça. Si le MOSFET reste un peu passant en permanence, cela peut expliquer le problème. En déconnectant la sortie et la TIC, je devrai avoir 0V sur la gate et 3,3V en sortie, et en réalité j'ai 0,22V sur la gate et 1,14V sur la sortie, donc le MOSFET conduit alors qu'il ne devrait pas.

En tous cas la bibliothèque marche super une fois le signal correct, et elle est très proprement faite (je suis connaisseur...).

hallard, 2024-07-19:
Possible que ce soit le mosfet (ou une mauvaise soudure sur une de ses pattes) mais c'est la 1ere fois que ça arrive.

Je peux vous envoyer un mosfet si vous voulez tenter la chose ou sinon je vous change la carte ?

Merci, c'est vrai qu'elle date cette librairie mais j'ai effectivement essayé de faire propre à l'époque :-)

Gixy31, 2024-07-19:
Je vais regarder les soudures au binoculaire et je vous recontacte. J'ai ce qu'il faut pour dessouder/ressouder, on va essayer de faire au mieux pour éviter le retour de la carte.

hallard, 2024-07-19:
ok, je pense que n'importe quel FET N fera l'affaire

Gixy31, 2024-07-19:
La soudure de la broche source, placée sur un via à la masse était suspecte. J'ai refais les trois et le seul changement est qu'au lieu de 0,2V sur la gate, j'ai maintenant 0,3V et toujours 1,16V en sortie, donc toujours pas glop. Je trouve aussi la soudure de la LED bleue pas terrible, elle aussi avec un via sur un pad. Je mesure 10k dans les deux sens à ses bornes, ce qui n'est pas normal. J'aurais donc tendance à préférer une autre carte car ce n'est peut-être pas le MOSFET. Si je dois te la renvoyer, il faut que je dessoude les barrettes avant.

hallard, 2024-07-19:
Non tu peux laisser les barrettes en revanche je la testerais ici aussi pour voir :-) 

En revanche je comprends pas y'a pas de via autours de la led bleu je viens de vérifier tu aurais une photo HD a mettre ?

Gixy31, 2024-07-19:
Exact, c'était une impression. Mais à y regarder de près, cette LED est plus collée que soudée, elle n'est pas d'aplomb, avec deux gouttes de soudure pas étalée en dessous. Problème, je ne peux plus couper la trace qui l'inhibe, puisqu'elle est sous le connecteur :-(

Gixy31, 2024-07-19:
Et pour les 11k constaté aux bornes, c'est normal, on passe par le 3,3V et on fait le tour par R5 et R6...

Gixy31, 2024-07-19:
En revanche, tout débranché, j'ai 4,7k entre drain et source, dans les deux sens, pas normal ça.

Gixy31, 2024-07-23:
Hello,
Sommes nous restés entendus sur le fait que tu m'envoies un MOSFET ?
L'appli est fonctionnelle, j'ai toutes les infos sur Grafana, reste à peaufiner les notifications, display local, etc.

hallard, 2024-07-23:
Yes, tu peux me donner ton numéro de commande Tindie, je t'envois à la même adresse (si d'ici 10min ça part today)

Gixy31, 2024-08-01:
Hello,
N'ayant toujours rien reçu, j'ai fini par cannibaliser un MOSFET sur une vieille carte qui trainait par là et hop, ça marche, c'était  donc bien le MOSFET qui dysfonctionnait. Les données captées par un ESP32, transmises à mon broker MQTT puis à ma base InfluxDB et enfin affichées dans Grafana. Ça donne ceci : 
[image]

hallard, 2024-08-02:
Top, merci pour le retour, j'attendais ton numéro d'order tindie pour retrouver ton adresse pour t'envoyer le mosfet (offre tjs valable bien sur)

Gixy31, 2024-08-02:
J'avais envoyé le n° de commande en répondant au mail, manifestement ça n'est pas passé. Ça marche comme ça, donc ce n'est plus la peine, merci.

closed 2025-01-08 by hallard