hallard/WeMos-TIC #18: choix de la board

sinclairfr, 2024-10-11:
Bonjour à tous,
Doit-on s'attendre une différence significative en termes de "performance" selon si on choisit ESP8266 ou ESP32 ?
Il me semble qu'un ESP8266 est largement suffisant mais je préfère demander avant de souder la board.
Merci !

hallard, 2024-10-11:
Le jour et la nuit, ne serait-ce que 

- pour la console série impossible (ou du moins compliqué) avec le 8266 car un seul port série
- MQTT over TLS (pas assez de RAM)
- Pouvoir faire du Berry avec Tasmota
- Vitesse, RAM, ...
- plus d'IO
- support du micropython
- Et j'en passe

En plus c'est quasiment le même prix, aucune hésitation

sinclairfr, 2024-10-12:
Effectivement, c'est le même prix et j'ai les deux donc va pour le ESP32. Merci !

closed 2025-01-08 by hallard