hallard/WeMos-TIC #19: WeMos-TIC + Tasmota : Bug depuis passage en mode standard

ronan-fontenay, 2025-01-08:
Bonjour,

**1 - Matériel**
- Wemos D1 ESP8266
- Wemos Teleinfo v1.2.
- 
**2 - Logiciel**
Tasmota 14.4.1.1 (compilé avec la Teleinfo).

**3 - Mode Historique **
Parfait fonctionnement. 

**4 - Mode Standard **
Demande à Enedis de passer en mode standard, depuis impossible d'avoir un module qui fonctionne. 

Etapes réalisées et problèmes constatés : 

1 - Reset Tasmota (pour partir d'un état initial) : OK

2 - Configuration au Wifi : OK

3 - Template : OK
{"NAME":"Wemos Teleinfo","GPIO":[1,1,1,1,640,608,1,1,1,5152,1376,1,1,1],"FLAG":0,"BASE":18}
[image]

4 - Reboot : OK 
Pas d'info car Linky en standard et Tasmota en Historique, a ce stade tout semble OK.
[image]
A ce stade, voici la configuration : 
16:14:40.626 CMD: EnergyConfig
16:14:40.630 TIC: Settings Mode:historique, RX:GPIO13, EN:None, Raw:noraw, Skip:0, Limit:0, Stats:0
16:14:40.633 RSL: RESULT = {"EnergyConfig":"Done"}

5 - Configuration en "standard
16:15:23.921 CMD: EnergyConfig Standard
16:15:23.925 TIC: 'standard' mode
16:15:23.926 TIC: RX on GPIO13, baudrate 9600
16:15:23.936 SNS: Hardware Serial
16:15:23.937 TIC: using hardware serial
16:15:23.940 RSL: RESULT = {"EnergyConfig":"Done"}

Et là, après reboot c'est la cata :
Page principale : 
[image]
Configuration perdue : 
[image]
16:18:17.613 CMD: energyconfig
16:18:17.619 RSL: RESULT = {"Command":"Unknown","Input":"ENERGYCONFIG"}

ronan-fontenay, 2025-01-08:
Bonjour, 
Merci pour cette réponse rapide ! 
Je viens d'essayer sans succès. 

J'ai branché la D1 sur le PC, et après activation du template, j'ai l'impression qu'elle redémarre plusieurs fois. 
`00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F7, Count 37

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F7, Count 37

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F7, Count 37

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F7, Count 37

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F7, Count 37`

Je viens de commander une D1 en ESP32 en me disant que c'est peut-être un problème de mémoire. 
Il me semble être tombé sur un sujet du forum dans lequel la personne avait réussi en tasmota V12 mais plus après. Je pourrais essayer un downgrade. Est-ce une idée ?

ronan-fontenay, 2025-01-08:
Flashé le firmware de @NicolasBernaerts avec succès mais j'ai exactement le même problème. 

La configuration se passe bien, mais dès que je coche "standard", je perds toute la conf après le reboot.
[image]

Je vais attendre l'ESP32 qui arrive vendredi :). 
Si d'autres reproduisent le problème, on pourrait peut-être mettre à jour la doc en conseillant fortement l'ESP32 ?

hallard, 2025-01-08:
je le crie à tou va partout :-)

[image]

ronan-fontenay, 2025-01-08:
Oups la honte, je n'avais pas vu le paragraphe. 
Merci d'avoir pris le temps de me répondre, et merci pour ton excellent travail :+1:

NicolasBernaerts, 2025-01-08:
Bonjour, 

Vous avez sans doute un WeMos D1 duplicate. Il y a en a beaucoup en circulation. Ils deviennent totalement instables dès qu'ils sont trop chargés. 

C'est vrai avec beaucoup de firmwares.

Je ne sais pas pourquoi ...

hallard, 2025-01-08:
Peut être avec un régulateur 3.3V de merde on avait le même soucis avec les brownout a répétition sur les D1 ESP32 aussi

ronan-fontenay, 2025-01-08:
Je vous ferai un retour après passage à l'ESP32 (une AzDelivery depuis leur shop officiel). 
Merci pour vos feedbacks ! Je ne voyais pas autant de rebondissements pour un simple changement de mode TIC :)

hallard, 2025-01-08:
> Je ne voyais pas autant de rebondissements pour un simple changement de mode TIC :)

Et moi donc, à ajuster les valeurs de résistances pendant 2 ans pour que ça marche sur tous les compteurs et quand arrive le mode standard qui fou tout par terre ...

ronan-fontenay, 2025-01-18:
Fonctionne parfaitement avec ESP32 👍 

[image]

hallard, 2025-01-18:
Nickel, je te conseille l'excellent firmware de Nicolas :-)

https://github.com/NicolasBernaerts/tasmota/tree/master/teleinfo/binary